enjoy-digital/litepcie #2: Confused on setup.

omstation, 2017-08-02:
Firstly is it possible to use this core with this board:
http://store.digilentinc.com/arty-artix-7-fpga-development-board-for-makers-and-hobbyists/

I see that litex should support it.

Also the setup instructions confuse me, as I have never worked with this kind of stuff before.

How could one use this with vivado for example? Does it produce IP that can be imported.

Sorry for my confusion as I am new to the subject.

mithro, 2017-08-02:
LitePCIe requires a board which has high speed transceivers on a PCIe connector. The Arty does not have this.

omstation, 2017-08-02:
@mithro Would it be possible to do PCIE using an Arty. Or is the high speed transceivers a necessity.

mithro, 2017-08-02:
High speed transceivers are a necessity.

omstation, 2017-08-02:
Just to reiterate, it is impossible to do PCIE (not specifically litepcie) with an Arty? Want to make sure I expressed that question correctly before. Are you aware of any boards that could support Lite PCIE, under a couple hundred dollars?

mithro, 2017-08-02:
It is *impossible* to do PCIE (not just LitePCIe) on the Arty without some type of expansion board with a PCIe SERDES (I have never seen such a thing and it would be quite expensive).

I've never found or used a PCIE compatible dev board under a couple hundred USD out there (definitely never seen a Xilinx based ones). 

The closest is possible something based on the Lattice ECP5 / ECP3 like http://www.latticesemi.com/en/Products/DevelopmentBoardsAndKits/ECP5VersaDevelopmentKit.aspx -- I don't believe that LitePCI currently support that board however. There was some recent developments around adding support for some Lattice boards but I don't know which ones.

BTW I'm not the developer of LitePCIe, I have just employed @enjoy-digital to do development for me and been very happy with the results that I now work with him regularly. Normally he would respond to this request reasonable quickly but he is currently on holidays with limited net access, so it might be a week or two before he responds with more information.

I'm actually working on a project to make an open hardware Artix-7 based board with PCIe which will hopefully be sub $300 USD. Progress is slow but steady -- hoping to see a release by the end of the year.

BTW If you want to chat in real time, you can join myself on [irc://irc.freenode.net/#timvideos or irc://irc.freenode.net/#m-labs](http://webchat.freenode.net/?channels=#timvideos,#m-labs).

closed 2017-11-06 by enjoy-digital