enjoy-digital/litepcie #8: conflict IRQ

ShawnLess, 2018-01-31:
Hi guys:

I tried to run the example but ran into following errors. Any suggestion?

## Problem
```bash
[shaolin@shaolin example_designs]$ ./make.py all
Traceback (most recent call last):
  File "./make.py", line 92, in <module>
    soc = top_class(platform, **top_kwargs)
  File "/home/shaolin/Lattice/litepcie/example_designs/targets/dma.py", line 61, in __init__
    with_timer=False
  File "/usr/local/lib/python3.5/site-packages/litex-0.1-py3.5.egg/litex/soc/integration/soc_core.py", line 176, in __init__
    interrupt, mod_name, interrupt_rmap[interrupt]))
```
## My configuration
1.  python 3.5
2.  litex version
```bash
[shaolin@shaolin litex]$ git show
commit 2ecd1b06668ff4d896e9462f2e967de783fb9792
Merge: c83ae98 0ac3530
Author: enjoy-digital <florent@enjoy-digital.fr>
Date:   Fri Jan 26 01:58:37 2018 +0100

    Merge pull request #61 from PaulSchulz/master
```
3. litedram version
```bash
[shaolin@shaolin litedram]$ git show
commit ec9ad2fc3905e9dae86b5f3c2bf243dd5ee5087f
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Wed Jan 31 09:32:21 2018 +0100

    frontend/dma: add description of fifo_buffered parameter
```
4. litepcie version
```bash
[shaolin@shaolin litepcie]$ git show
commit 7b5b8063b3beaf76a12dcdda87dfcf91b1b79707
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Sun Jan 28 03:04:49 2018 +0100

    core/tlp/depacketizer: simplify using NextValue
```

enjoy-digital, 2018-01-31:
Hello,
it should be fixed, can you retry?

ShawnLess, 2018-01-31:
Thanks!  Thought there are other errors (I need to install vivado), but previous error disappeared ...

enjoy-digital, 2018-02-01:
Hi, yes vivado is required to impement this example design, so i'm closing this.

closed 2018-02-01 by enjoy-digital