enjoy-digital/litepcie #9: Is there anyway to specify the install dir of the FPGA tools?

ShawnLess, 2018-02-01:
Hey guys:
Nice work but I still ran into problem after I installed the Xilinx tools:
```bash
-bash-4.1$ ./make.py all

      __   _ __      ___  _________
     / /  (_) /____ / _ \/ ___/  _/__
    / /__/ / __/ -_) ___/ /___/ // -_)
   /____/_/\__/\__/_/   \___/___/\__/

  A small footprint and configurable PCIe
          core powered by Migen
====== Building options: ======
Platform:  kc705
Target:    dma
Subtarget: PCIeDMASoC
System Clk: 125.0 MHz
===============================
Traceback (most recent call last):
  File "./make.py", line 154, in <module>
    vns = platform.build(soc, build_name=build_name, **build_kwargs)
  File "/homes/shx089/tools/lib/python3.5/site-packages/litex-0.1-py3.5.egg/litex/build/xilinx/platform.py", line 33, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/homes/shx089/tools/lib/python3.5/site-packages/litex-0.1-py3.5.egg/litex/build/xilinx/vivado.py", line 204, in build
    _run_vivado(build_name, toolchain_path, source)
  File "/homes/shx089/tools/lib/python3.5/site-packages/litex-0.1-py3.5.egg/litex/build/xilinx/vivado.py", line 65, in _run_vivado
    settings = common.settings(vivado_path, "Vivado", ver, first="name")
  File "/homes/shx089/tools/lib/python3.5/site-packages/litex-0.1-py3.5.egg/litex/build/xilinx/common.py", line 51, in settings
    ver = max(vers)
  File "/homes/shx089/tools/lib/python3.5/site-packages/litex-0.1-py3.5.egg/litex/build/tools.py", line 34, in versions
    for n in os.listdir(path):
FileNotFoundError: [Errno 2] No such file or directory: '/opt/Xilinx/Vivado
```
I installed the vivado in a dir other than **/opt/Xilinx/Vivado**. How can I change that? I there any parameter that I can override on command line? 

Thanks in advance.

enjoy-digital, 2018-02-01:
You can try to use toolchain_path="your_toolchain_path" here:https://github.com/enjoy-digital/litepcie/blob/master/example_designs/make.py#L154

enjoy-digital, 2018-02-01:
Note that if you want to build this example design, you will need to have a valid license for a xc7k325t. I should add an example design for artix7 too.

ShawnLess, 2018-02-01:
Hi :

Thanks for the kind reminder.  I will check if we have that license feature. 

Anywhere, is it possible to generated RTL only? we don’t need bitstream at this stage. 

Regards,
Shaolin.

ShawnLess, 2018-02-01:
Adding ** toolchain_path="your_toolchain_path"** to: https://github.com/enjoy-digital/litepcie/blob/master/example_designs/make.py#L154 worked perfectly.

Anywhere, I lack the license to generate the bitstream.

enjoy-digital, 2018-02-09:
Closing since solved.

closed 2018-02-09 by enjoy-digital