enjoy-digital/litepcie #11: Add Power9 support

enjoy-digital, 2018-08-07:
Quick test on Power9 with PCIe X4 / Gen2 report issue:
` EEH: PHB#0 failure detected, location: N/A`
Probably related to the kernel driver.

mithro, 2019-05-16:
@shenki - Any chance you could help @enjoy-digital (or find someone who could help) with getting LitePCIe working correctly on power hardware?

enjoy-digital, 2019-05-17:
I still have the machine, so can do test, investigate on it.

shenki, 2019-05-21:
Yes, I can find folks who can help.

What do you have attached? If I can program a FPGA board with the same gateware I can help debug.

enjoy-digital, 2019-05-21:
@shenki: great thanks. I tested that a long time ago, so i'll first reproduce in the next days to give more information. I was testing with an Artix7 board, do you have access to an Artix7/Kintex7 FPGA PCIe board? If yes, can you tell me which one? i could produce a bitstream for you so that we both reproduce the behaviour.

shenki, 2019-05-22:
As it turns out I don't have any PCIe capable FPGA boards.

Can you provide some more information about the failure?

Kernel version: `uname -a`
skiboot logs: `cat /sys/firmware/opal/msglog`
kernel log: `dmesg`
PCI information: `lspci -vvv`

If your kernel is crashing, are you able to get the above information from the peteitboot environment?

mithro, 2019-05-22:
[A NeTV2](https://www.crowdsupply.com/alphamax/netv2) is potentially a good option?

oohal, 2019-05-22:
> Quick test on Power9 with PCIe X4 / Gen2 report issue:
> ` EEH: PHB#0 failure detected, location: N/A`
> Probably related to the kernel driver.

EEH events get invoked for a lot of reasons, but it's usually driver bugs (bad MMIO loads, forgetting to setup a DMA map, etc), but it's also possible there's a bug in your PCIe logic. We'd need the rest of the EEH register dump to work out the actual problem.

> @shenki: great thanks. I tested that a long time ago, so i'll first reproduce in the next days to give more information. I was testing with an Artix7 board, do you have access to an Artix7/Kintex7 [...]

Looks like the only FPGA card we have on hand is [this thing](http://www.nallatech.com/wp-content/uploads/pcie_385pb_v1_21.pdf) which is Stratix-V based and doesn't appear to have any real documentation available.

mithro, 2019-05-22:
I could lend someone a NeTV2 card if it would be useful? I believe @enjoy-digital has one too?

mikey, 2019-05-22:
> I could lend someone a NeTV2 card if it would be useful? I believe @enjoy-digital has one too?

Sure, we can look at a card in Canberra or you can send it to @shenki in Adelaide.

Is there a Linux kernel driver for this card? If so, where is it?

enjoy-digital, 2019-05-22:
Thanks all the the help. I don't want you too much time on it before i do it myself :) As a first test, i'll reproduce the behaviour, will try to give you more logs/reports and will create a minimal design to share with you the Linux driver code. Then if we still don't explain the behavior, we'll see how you can reproduce.

Note: since the first time i investigated, i added Altera Cyclone V support to LitePCIe, so one of the test i also want to do is testing with an Altera board (i only tested with a Xilinx board before). This will help to see if it could be related to the hardware or driver since the Hard IP in Altera and Xilinx FPGA are not the same.

oohal, 2019-07-24:
ping?

enjoy-digital, 2019-07-24:
@oohal sorry, i've not tested again, will try to do it soon.

enjoy-digital, 2021-01-08:
Closing since we don't have a real need for it, that was just nice to have. This is probably just some minor adaptations to do on the kernel driver that we could look at if we have a need to support Power9.

closed 2021-01-08 by enjoy-digital