enjoy-digital/litepcie #14: Possible roadmap for the project for PCIE Gen3?

anmolsahoo25, 2018-11-09:
Would it be possible for the lead developer to share a possible guideline doc for upgrading this library to support PCIE Gen3?

Interested in contributing to the development and I have started reading the source code as well, but it would be nice if the lead developer could share their insight.

Upto the developer though, if possible!

enjoy-digital, 2018-11-09:
Hi,

PCIe gen3 support is planned and that would i would be really interested in helping you getting that working if you want to contribute.

The core is already supporting PCIe up to X4 in gen2, so it should support PCIe gen3 up to X2 without modifications.

To add PCIe gen3 support, you could:
- Make sure to have a FPGA board that supports PCIe gen2 and PCIe gen3. (Xilinx KC705 would be perfect for that).
- Test the example design in PCIe gen2.
- Create a PCIe gen3 PHY with the Xilinx wizard.
- Create a wrapper (the interface should be very similar the the PCIe gen2 PHY).
- Test :)

Which PCIe board do you have? I you are using a board i also have (KC705/KCU105), i could help debugging.

enjoy-digital, 2018-11-09:
Thanks, yes please share your progress, results. You should probably start with the VC707 which will be closer to the KC705 supported by the example.

anmolsahoo25, 2018-11-09:
Sure!

enjoy-digital, 2019-04-17:
Hi @anmolsahoo25,

have you been able to make progress? Are you still planning to do it?

enjoy-digital, 2019-06-17:
Closing since no update.

closed 2019-06-17 by enjoy-digital