enjoy-digital/litepcie #17: Segmentation fault when running litepcie_util dma_loopback_test

nanortemis, 2019-01-04:
Hi, 

I'm getting a segmentation fault when running `litepcie_util dma_loopback_test`

lspci output
```
01:00.0 Memory controller: Xilinx Corporation Device 7022 (rev 01)
        Subsystem: Xilinx Corporation Device 0007
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Region 0: Memory at f7c00000 (32-bit, non-prefetchable) [size=1M]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit-
                Address: 00000000  Data: 0000
        Capabilities: [60] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0, Latency L0s <64ns, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 75.000W
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 5GT/s, Width x2, ASPM L0s, Exit Latency L0s unlimited, L1 unlimited
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range B, TimeoutDis-, LTR-, OBFF Not Supported
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v1] Device Serial Number 00-00-00-01-01-00-0a-35
```

dmesg output after calling ./init.sh
```
[  514.864422] litepcie: loading out-of-tree module taints kernel.
[  514.864484] litepcie: module verification failed: signature and/or required key missing - tainting kernel
[  514.864868] litepcie Probing device
[  514.865949] litepcie Assigned to minor 0
```
output after calling ./litepcie_util dma_loopback_test
```
./litepcie_util dma_loopback_test
Segmentation fault (core dumped)
```

dmesg
```
[  674.303697] show_signal_msg: 35 callbacks suppressed
[  674.303699] litepcie_util[28008]: segfault at 7fa8a4c89000 ip 00005640ca618450 sp 00007ffdaffffa00 error 6 in litepcie_util[5640ca618000+1000]
[  674.303705] Code: 01 44 24 14 c1 ee 02 48 63 c9 74 33 69 54 24 0c 76 59 41 31 44 8d 46 ff 48 8d 04 0f 4a 8d 7c 87 04 48 01 f9 83 c2 01 0f 1f 00 <89> 10 48 83 c0 04 81 c2 76 59 41 31 48 39 c1 75 ef 01 74 24 0c 8b
```

nanortemis, 2019-01-15:
any findings? I can do the testing...

enjoy-digital, 2019-01-15:
Sorry, i just tested and fixed with https://github.com/enjoy-digital/litepcie/commit/324ab72b895547056bff68d8a3293e184624981.

closed 2019-01-15 by enjoy-digital

nanortemis, 2019-01-16:
Thanks it works.