enjoy-digital/litepcie #34: Allow MMAP access to BAR0 without loading kernel driver

enjoy-digital, 2020-07-02:
Once the driver is loaded, it's possible to access BAR0 to access the wishbone bridge of the SoC to access registers or create a virtual console:

````
# Create bridge:
# --------------
# sudo litex_server --pcie --pcie-bar=/sys/bus/pci/devices/0000\:01\:00.0/resource0

# Use scripts:
# -------------
# ./test_regs.py
# ./litescope_wishbone.py (--help to see preconfigured triggers)

# Use console:
# litex_crossover_uart --base-address=-0x82000000 (will create /dev/pts/X)
# litex_term /dev/ptsX
````

If possible, we should allow this without loading the kernel driver and understand what needs to be enabled.

zyp, 2020-07-02:
I've found that if I bind the `uio_pci_generic` driver to the device, I'm allowed to access BAR0 in the same way.

enjoy-digital, 2020-07-02:
@zyp: thanks! I just also tested it successfully with:
````
sudo modprobe uio_pci_generic
sudo chmod 777 /sys/bus/pci/drivers/uio_pci_generic/new_id
echo "10ee 7024" > /sys/bus/pci/drivers/uio_pci_generic/new_id
sudo litex_server --pcie --pcie-bar=/sys/bus/pci/devices/0000\:02\:00.0/resource0
````

Then using scripts similar to https://github.com/enjoy-digital/litex/wiki/Use-Host-Bridge-to-control-debug-a-SoC#create-scripts-to-communicate-from-the-host-pc-with-the-bridge.

I will add it the the wiki.

ozbenh, 2020-07-03:
You could just poke the enable file in sysfs no ? That said I would recommend using uio APIs or vfio for cleanliness. VFIO also handles mapping your userspace address space in the iommu for DMA which is nice

zyp, 2020-07-03:
Yeah, you're right, just tested that. Didn't know it were that simple. :)

enjoy-digital, 2021-01-08:
This is now integrated in CommPCIe: https://github.com/enjoy-digital/litex/blob/master/litex/tools/remote/comm_pcie.py#L25-L31. Running `litex_server --pcie --pcie-bar=xyyy` will automatically set the enable bit if not already set.

closed 2021-01-08 by enjoy-digital