enjoy-digital/litepcie #36: Check PCIe bus rescan after loading bitstream.

enjoy-digital, 2020-07-02:
It seems to be possible: https://stackoverflow.com/questions/32334870/how-to-do-a-true-rescan-of-pcie-bus but hasn't been tested.

zyp, 2020-07-02:
I've been doing this with the CLE-215+, works for me.

enjoy-digital, 2020-07-02:
@zyp thanks, indeed i also just tested it successfully on Ubuntu 18.04:
````
sudo chmod 777 /sys/bus/pci/devices/0000:02:00.0/remove
echo 1 /sys/bus/pci/devices/0000\:02\:00.0/remove
sudo chmod 777 /sys/bus/pci/rescan
sudo echo 1 > /sys/bus/pci/rescan
````

racerxdl, 2021-06-18:
@enjoy-digital I don't think it works on all machines though. Here the BAR doesn't get remmaped if I don't reboot. The device correctly get removed and when I issue rescan, they show up again, but it can't be used.

Here is a HP Server that I'm using to test it.

smunaut, 2022-04-26:
In case anyone is still looking for such info : doing a `remove` on the root bridge corresponding to that PCIe slot and then doing a rescan often works better because it forces reconfiguration of the memory zones being mapped / enabled through it.

racerxdl, 2022-06-23:
Just confirmed that it does work issuing a remove as @smunaut pointed out.

Just need to be carefull for Allocated BAR size, which kernel doesnt always find a place to fit it. But it does correctly accept a new device.