enjoy-digital/litepcie #38: Queries about litepcie

imBilal1998, 2020-08-07:
Hi!

While I was searching for a xilinx FPGA solution for PCI Express, I stumbled upon litepcie. Since there is very little documentation is available about the litepcie, I am writing my queries here. Other users are also welcomed to use this thread/issue to discuss their queries about litepcie. 

As stated on [litepcie page](https://github.com/enjoy-digital/litepcie), that it supports several xilinx devices such as ultrascale plus, ultrascale, 7 series etc. This leaves room for a lot of uncertainty. Such as, many ultrascale plus devices have [Ultrascale+ PCI Express Integrated Block](https://www.xilinx.com/support/documentation/ip_documentation/pcie4_uscale_plus/v1_3/pg213-pcie4-ultrascale-plus.pdf) which include **ZCU106 Board(part:xczu7ev-ffvc1156-2)** whereas **ZCU102 Board(part:xczu9eg-ffvc1156-2)** does not have any integrated PCIe block and it only uses the [PCI PHY 1.0](https://www.xilinx.com/support/documentation/ip_documentation/pcie_phy/v1_0/pg239-pcie-phy.pdf).
 
So my point is, there are different version of PHY in different devices of the same category(e.g. ultrascale+). Now I am confused that which one is supported by litepcie? My best guess is that litepcie is build only for **Ultrascale+ PCI Express Integrated Block** so it will work on ZCU106 board but it will not work on ZCU102 board because there is no Integrated Block there. 

And also I want to know if there are any video/pdf tutorials about litepcie? 

One more thing: Under the getting started heading at litepcie page, step 3 needs more description. 
**TODO: add/describe examples**
What does this step 3 means?

Please clear the query. 
Bilal

enjoy-digital, 2020-08-07:
Hi @imBilal1998,

the documentation is indeed minimal for now (and we are going to improve this over time), but there are already several example of LitePCie integration on various boards:
-https://github.com/enjoy-digital/litepcie/tree/master/examples
-https://github.com/litex-hub/litex-boards/blob/master/README.md

Since LitePCIe currently relies on the integrated PCIe hard block of the supported devices, that's a requirement to get it working. Supporting others PCIe PHY would be possible but this is extra-work that has not been done yet (since we haven't had yet a use case for it).

A good board to start with LitePCIe is the Acorn CLE215+, the process to operate LitePCIe on is is explained in the target file:
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/acorn_cle_215.py#L6-L21

Hoping it answers your questions,

Florent

imBilal1998, 2020-08-07:
Thanks @enjoy-digital for such a quick reply. I would wish to keep this thread open for further queries for future users.

That very well answers my basic question. But can you tell me, on litepcie page **getting started** paragraph, What does the step 3 means? 

Also that on the litepcie page. it is mentioned that **LitePCIe can be used as LiteX library or can be integrated with your standard design flow by generating the verilog rtl that you will use as a standard core.** How can I generate the verilog rtl? On [litepcie phy folder](https://github.com/enjoy-digital/litepcie/tree/master/litepcie/phy/xilinx_usp_x4) some verilog are provided? Is it the already generated verilog rtl?

Current I don't have **Acorn CLE215** available and I am working on **ZCU102** board. So I want to know, how much work is required to make PCIe run on this board? How can I manually integrate litepcie with ZCU102 board, given that it does not have any integrated block? 

Thanks.

--Bilal

enjoy-digital, 2020-08-07:
The getting started is not yet written, the best for now is to look at the example design i was listing. You can use LitePCIe in a LiteX design by importing the LitePCIe core (full LiteX flow), but you can also generate a standalone LitePCIe core that you would just use as a verilog code in a traditional flow: for example in `examples` directory, you can generate a standalone core for the AC701 board with: `litepcie_gen ac701.yml` and then integrate in in your design using the corresponding provided `.xci`.

Regarding the ZCU102, it will provide the Physical Layer but the Data Link Layer will be missing since LitePCIe operates at the TLP layer, so it's possible, but would still requires some work (at least a few weeks full time for an experienced FPGA developer). If you want to work on this and contribute this, happy to help you and give directions, but otherwise, i would recommend using an FPGA with a PCIe hard block or use another core that would provide you all the layers (if available).

enjoy-digital, 2021-01-08:
Closing since the questions have been answered.

closed 2021-01-08 by enjoy-digital