enjoy-digital/litepcie #42: Move Ultrascale(+) TLP adaption code from Verilog to Migen

enjoy-digital, 2020-09-23:
The AXI streams exposed by the on Ultrascale (+) PHYs are not standardized TLPs. To support the Ultrascale (+) PHYs in LitePCIe (that operates on standardized TLPs), the AXI streams have been adapted for both downstream/upstream directions in `pcie_us(p)_support.v` files. To simplify the code/integration and avoid too much code duplication, it would be good to move this code to Migen which would also ease supporting various data widths.

smunaut, 2022-03-01:
Would definitely like to see this done and could contribute to a bounty for it.

enjoy-digital, 2024-01-12:
A first step in this direction has been done with https://github.com/enjoy-digital/litepcie/pull/123, we'll now need to properly get rid of the verilog doing the adaptation between Xilinx's standard and PCIe's standard for the TLP streams.