enjoy-digital/litepcie #43: Is there any difference between vcu1525 and bcu1525?

YangWang92, 2020-11-20:
Hi, 
Thanks for your work on PCIe core, and I found that this core support bcu1525 (https://github.com/enjoy-digital/litepcie/blob/master/examples/xcu1525.py).

I cannot find any schematic about bcu1525, and does this board keep the same pinout as vcu1525? Should I modify some codes when porting the core from bcu1525 to vcu1525 or vice versa? 

Thanks a lot!

enjoy-digital, 2020-11-20:
Hi @YangWang92,

the BCU1525 and XCU1525 have the same pinout. The probable only difference is the type of DDR4 SODIMM mounted on it. You can find more information at: https://github.com/d953i/Custom_Part_Data_Files/tree/master/Boards

YangWang92, 2020-11-20:
Thanks for your reply, I also found the repo but it does not mention pcie/qsfp28 pinouts.

Thhans again!

closed 2020-11-20 by YangWang92

alexforencich, 2020-12-02:
The VCU1525, VCU1525, Alveo U200, and Alveo U250 are all basically pin-compatible, with the exception of the transceiver sites on the U250 having different names.

YangWang92, 2020-12-02:
Thanks for your reply, and I'm working on BCU1525 used for mining at a low price on eBay. I generated bitstreams of VCU1525 and programmed it to BCU1525. It cannot scan QSFP28 links on this board, and I'm trying to figure it how it works.

alexforencich, 2020-12-02:
I have a BCU1525 and a VCU1525, and that design works fine on both.  Also, define "scan."  Not sure what you mean by that.

alexforencich, 2020-12-02:
Also, if this is a corundum issue, then maybe it would make sense to open an issue about on that repo.

YangWang92, 2020-12-02:
> I have a BCU1525 and a VCU1525, and that design works fine on both. Also, define "scan." Not sure what you mean by that.

I scanned QSFP28 ports with IBERT in Vivado, and I'm not sure whether is related to misconfiguration or hardware problems.

I will double-check it and update in corundum's repo.