enjoy-digital/litepcie #45: will litepcie be helpful for developing PCIE VIP

simdream, 2020-11-25:
For developing inhouse PCIE VIP, will this litepcie project be helpful as a start point? I assume the generated PCIE IP could be at least the BFM part, and on top we can build up SV UVM layers? Thanks,

enjoy-digital, 2020-12-03:
Hi,

the current version of LitePCIe is relying on hard PCIe block, so starts at the TLP level. So will handle MMAP and DMAs. You could indeed probably use LitePCIe to do some checks with your PCIe VIP, but it will not cover all the layers. For this, you could generate the LitePCIe standalone core with `litepcie_gen ac701.yml` (that will also integrate the Xilinx hard PCIe block) and simulate it with your PCIe VIP.

Regards,

Florent

simdream, 2020-12-03:
thanks Florent

closed 2020-12-03 by simdream