enjoy-digital/litepcie #50: I found no way in device driver to specify the starting address of DMA?

zhangzq71, 2021-03-03:
I can't find a statement in litepcie_dma_writer_start or litepcie_dma_reader_start to specify the starting address of each DMA operation. Is is necessary to set the start address?

enjoy-digital, 2021-03-03:
Hi @zhangzq71,

the current driver allocates the DMA buffers in `litepcie_pci_probe` and then use this to configure the DMA descriptor on the hardware in `litepcie_dma_writer_start` and `litepcie_dma_reader_start`:
- https://github.com/enjoy-digital/litepcie/blob/master/litepcie/software/kernel/main.c#L237-L238
- https://github.com/enjoy-digital/litepcie/blob/master/litepcie/software/kernel/main.c#L288-L289

So currently the memory allocation is handled by the driver. If you want to do DMA on specific addresses, you'll need to program the DMA descriptors with a customized function.

zhangzq71, 2021-03-03:
Hi,

The descriptor of SGDMA controller was configured in litepcie_dma_writer_start and litepcie_dma_readerer_start,
    for(i = 0; i < DMA_BUFFER_COUNT; i++) {
        litepcie_writel(s, dmachan->base + PCIE_DMA_WRITER_TABLE_VALUE_OFFSET,
#ifndef DMA_BUFFER_ALIGNED
                    DMA_LAST_DISABLE |
#endif
                    (!(i%DMA_BUFFER_PER_IRQ == 0)) * DMA_IRQ_DISABLE | /* generate an msi */
                    DMA_BUFFER_SIZE);                                  /* every n buffers */
        litepcie_writel(s, dmachan->base + PCIE_DMA_WRITER_TABLE_VALUE_OFFSET + 4,
                    dmachan->writer_handle[i]);
        litepcie_writel(s, dmachan->base + PCIE_DMA_WRITER_TABLE_WE_OFFSET, 1);
    }
for these code, dmachan->writer_handle[i] is the destination address that located in host, but I can't find where to set the source address that located in FPGA PCIE board, so I want to know what is the source address.

ZhangZQ

  *   https://github.com/enjoy-digital/litepcie/blob/master/litepcie/software/kernel/main.c#L237-L238
  *   https://github.com/enjoy-digital/litepcie/blob/master/litepcie/software/kernel/main.c#L288-L289

enjoy-digital, 2021-03-03:
The DMAs in LitePCIe are generating/receiving streams of data (similar to AXI Streams). So on the FPGA side, there are no MMAP/destination address, we only have DMAs channels with data sink and source. If you want to also do MMAP on the FPGA, you then have to add a stream to/from MMAP DMA and configure it over PCIe. Such modules can be the `WishboneDMAReader`/`WishboneDMAWriter` from https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/dma.py if you want to do DMA on the main bus of the SoC, or` LiteDRAMDMAReader``/LiteDRAMDMAWriter` if you want to do DMA directly to/from the DRAM memory.

zhangzq71, 2021-03-03:
Oh, I see, thank your very much!

ZhangZQ

closed 2021-05-03 by enjoy-digital