enjoy-digital/litepcie #51: A LitePCIeDMA class question

jimmymagemtek, 2021-07-30:
Hi Sir,
I know that the Host is just advertised by separate DMAReader/DMAWriter IRQs of the progress/status on the FPGA.

Then, how does the host ensure that writes are done ahead of FPGA reads (DMA Reader, Host --> FPGA) and reads are done after FPGA writes (DMAWriter, FPGA --> Host) ? 

Since the host seems to have no way to let FPGA to follow its write or wait for its read.

Please advise. Many thanks.
Jimmy

enjoy-digital, 2021-08-24:
Hi @jimmymagemtek,

In LitePCIe, the FPGA is the one maintaining the Scatter-Gather loop indexes and regulating the transfers. The Host is just advertised by separate DMAReader/DMAWriter IRQs of the progress/status on the FPGA and has to ensure that writes are done ahead of FPGA reads (DMA Reader, Host --> FPGA) and reads are done after FPGA writes (DMAWriter, FPGA --> Host).

enjoy-digital, 2021-09-08:
Closing since question has been answered. @jimmymagemtek since you are using a custom design built for a partner, can you contact the partner if you need more support than the one I can provide here?

closed 2021-09-08 by enjoy-digital