enjoy-digital/litepcie #64: DMA ADDR of the Device

tongchen126, 2021-11-18:
Could anyone give me some intuition about the DMA, like the address of the device memory it is transferring from? I want to transfer data between the main memory of my development board(KC705) and the host PC memory, is it doable by litepcie?

enjoy-digital, 2021-11-19:
Hi @tongchen126,

the DMA on the FPGA side are streams of data(similar to AXI-streams) : A source for the DMA Reader, a Sink for the DMA Writer. If can convert to this MMAP by connecting the opposite DMA from LiteX to it: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/dma.py and program this DMA over the PCIe.

In there is a need to do MMAP (Host) to MMAP (FPGA) (or the opposite), we could eventually think about adding a similar module.

tongchen126, 2021-11-27:
I use the following code, wishing to transfer data between Host Memory(starting from 0x0000_0000) and FPGA Memory(Wishbone bus). This code is basically based on the one raised by [https://giters.com/enjoy-digital/litepcie/issues/54](url). However, when I read the Host Memory on the Litex command line. It doesn't give the desired data output(As the image shows, the physical memory of the host is on the left side, and the data read by the FPGA is on the right side). 
If you can give me any suggestions or hints?

`
from litex.soc.integration.soc import SoCRegion
from litex.soc.interconnect.axi import AXILiteInterface, AXI2AXILite,AXILite2AXI,Wishbone2AXI
from litepcie.frontend.axi import LitePCIeAXISlave

pcie_host_axi_mmap = LitePCIeAXISlave(self.pcie_endpoint, data_width=32)
pcie_host_wb = wishbone.Interface()
self.submodules += pcie_host_axi_mmap, Wishbone2AXI(pcie_host_wb,pcie_host_axi_mmap.axi,base_address=0x2000_0000)
self.bus.add_slave("pcie_host_axi", pcie_host_wb, SoCRegion(origin=0x2000_0000, size=0x1000_0000))
`
[image]

tongchen126, 2021-11-27:
I have do this successfully based on LitePCIeWishboneSlave. It gives the derived output.

from litex.soc.integration.soc import SoCRegion
from litepcie.frontend.wishbone import LitePCIeWishboneSlave
pcie_host_wb_mmap = LitePCIeWishboneSlave(endpoint)
self.submodules += pcie_host_wb_mmap
self.bus.add_slave("pcie_host_wb", pcie_host_wb_mmap.wishbone, SoCRegion(origin=0x2000_0000, size=0x1000_0000))  # 16MB.
(Also modified this [line](https://github.com/enjoy-digital/litepcie/blob/master/litepcie/frontend/wishbone.py#L129) to minus the base address (0x2000_0000 >> 2))

tongchen126, 2021-11-27:
And to mention that I didn't load the litepcie kernel module. The LitePCIeWishboneSlave works as soon as the PC boots(even before booting into kernel).

closed 2021-12-12 by tongchen126