enjoy-digital/litepcie #83: Support for Xilinx soft PCIe PHY?

Elgox2, 2022-01-05:
Hello,

I'm wondering if parts of this project can be used to control the Xilinx soft PCIe PHY ([PG239](https://www.xilinx.com/content/dam/xilinx/support/documentation/ip_documentation/pcie_phy/v1_0/pg239-pcie-phy.pdf))? If I'm right, the provided examples only use the hardened-in-silicon IPs from Xilinx. 

Any experience with that core, or do you known some other resources for that?

Thanks :)

enjoy-digital, 2022-01-23:
Hello @Elgox2,

LitePCIe is indeed using the hardened IPs with TLP interfaces. The Data Link and Transaction Layer will be missing. So this will not be directly supported but it could be a good intermediate step for a full open-source PCIe PHY. The Xilinx PHY could be reused + LitePCIe and the Data Link/Transaction Layers would be developed. Once working, we could also try to replace the PHY, this will probably consist of reusing LiteICLink wrappers + handling LTSSM.

JCBuck, 2022-01-26:
Hi @Elgox2 and @enjoy-digital, 

Perhaps this question is better suited for Xilinx themselves, but according to (PG239):
```haskell
The Vivado® IP catalog does not allow generation of this IP for all UltraScale™ and UltraScale+™
devices; however, if a device is selected and has the same transceiver type as the desired device
(UltraScale GTH, UltraScale+ GTH or UltraScale+ GTY), the IP can then be migrated to the
desired part.
Currently, the IP can be generated for the following devices:
• UltraScale+: ZU9EG (GTH), VU3P (GTY), and VU9P(GTY).
• UltraScale: KU040 (GTH), KU115 (GTH), VU440 (GTH), and VU440 ES2 (GTH)
```
I'm looking at a `Kintex-7 KU9P` which does not contain any `PCIE4` or `PCIE4C` blocks but does have `16Gb/s GTH transceivers` according to [ds890](https://www.xilinx.com/support/documentation/data_sheets/ds890-ultrascale-overview.pdf)

Does the verbiage in PG239 imply that the soft PHY could be modified to work with the `KU9P` variant and subsequently LicePCIe, if and when the remaining layers are implemented?

enjoy-digital, 2022-02-01:
Hi @JCBuck,

it seems the IP could indeed be generated for your device, but as I was describing in the previous message, it would still requires the Data Link/Transaction Layer to be implemented to work with LitePCIe.