enjoy-digital/litepcie #100: litecpie dma test result in error

1n3o1, 2022-09-03:
thanks in advance 
after building both fpga and kernel I tested the dma which resulted in error 
it will show error in dma test if I use it without any change to configs 
if I change its width(lite_util config (its 16bit width by default )) to 32 bit its gets better but still encounter error 
or if I change it to zero copy it gets resolved 
and also I have tested the old version for last year which had no zero copy 
in that version also there was moment which in dma test it outputs too many errors .

enjoy-digital, 2022-09-14:
Hi @1n3o1,

Can your provide the error message and configuration?

Also I would recommend reading https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support#feedback--contribution--support before creating an issue: If you don't share the useful information you have (here error message, specific config, etc...), we'll have to spend time trying to guess what you are doing differently.

1n3o1, 2022-09-18:
Hi 
first that new litepcie cant be built because the uart part is broken I had to modify the make file to exclude that 
there is a missing XA_LIMIT and xa_struct which I have no clue what they should be .
and also I did not mean an error code for dma test I mean the error counter inside dma test 
if I use anything other than zero copy dma error counter would go high and shows error on every transaction .
you can test it yourself just build a board with pcie dma and test it with hardware to see if dma test form litepcie utils works.
(I think there is something wrong with the new read and write file operation for pcie software)

enjoy-digital, 2022-09-18:
Sorry, I took time to write https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support, so please read it and think about your tone before submitting another issue.

closed 2022-09-18 by enjoy-digital