enjoy-digital/litepcie #101: there is no packet separation capability for dma

1n3o1, 2022-09-03:
thanks in advance 
I need to indicate start and end of packet for DMA 
but there is no indication in scatter gather table for sart and end of packet 
requesting addition of RX/TX SOF/EOF to the descriptors.
(tlast wont reach user I have to use escape coding to separate my packets or have a packet base transmission and use their header length to seperate data packets and create which it should be produced by DMA scatter gather descriptor  )

enjoy-digital, 2022-09-14:
The DMA are using LiteX's stream and last signal should be support for both DMAWriter/Reader. Please share more information if you want us to have a closer look.

1n3o1, 2022-09-18:
no actually there is no tlast.
of course that there exist a tlast signal within the interface but it connects to nowhere !!
I have followed both generated rtl and also its python code and I have confirmed that tlast dose not get transferred 
and if you think about the design you would understand that it is not possible to transfer tlast with in this architecture 
for being able to generate tlast your dma descriptors like xilinx dma has to have a new field called end of packet or start of packet that Identify packets inside stream .
that means after execution of descriptor there is no grantee how many SG list or descriptor has to be activated to generate packets
you can create tlast of user data side by tlast you get form request completion form pcie but as you know descriptors gets splitted  as a result even one descriptor can have multiple tlast. so 
the only solution is to add a new field to the descriptors called EOF SOF that signals the start tlast when this field is activated 
with that architecture user can send parts of a packet (like its header and body ) through SG list and only activate tlast for the last descriptors that actually has to have a tlast 
that was for dma reader .
but also for dma writer it should be possible to gets terminated if packet arrives even if the requested length not met . 
that means it has to discard  all of splitted descriptors and finish the transaction 
for that it has to have a mechanism to identify if a packet exits (the same way that you grantee that minimum amount of data is within the dma to write ) and then activate dma write and if tlast gets in it drops all the remaining splitted desc. and finishes this one with zero padding.(the length has already send through pcie tlp packets so either you have to add a buffer to capture 
packets and finds their length (that would be  hard ) or just zero pad until it reaches the length that it requested )
user has to find the length some other way like in the header of packet.
(it is not possible to forecast the length if user wants packets instead of streams he could use stream kind that write 
continuously but that way there is no grantee that the last packet will reach host (because the length could not fill the requested pcie packet ) )

enjoy-digital, 2022-09-18:
Sorry, I took time to write https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support, so please read it and think about your tone before submitting another issue.

closed 2022-09-18 by enjoy-digital