enjoy-digital/litepcie #98: clock domain separation of litepcie

1n3o1, 2022-09-03:
thanks in advance
the structure of litepcie is the way the it wont let you to separate the pcie dma and its related modules along TLP handler and endpoint form soc clock domain it will only let you to do CDC after endpoint which result in bandwidth loss
for instance if I use a 250mhz 64 bandwidth endpoint then I have to CDC it to CPU clock of 100 mhz that means more that half of 
DMA BW is dropped also there is no chance in increasing the soc cpu clock 
as you know clock frequency of CPU can be low as 100mhz but I need to have a high performance high throughput pcie dma winch is separate form CPU clock domain .
that means it has to have a CDC for its control form CPU and has a CDC over its CSR register 
but the problem here is that will causes a lot more resource utilization you have to CDC its wishbone and then use a CSR on its own clock that means it would be better to have a separate wishbone and CSR from SOC and then have CDC connection form soc or to soc (also I dont see any need for connection to soc only master of pcie lite (that is the PC) has to have access to these CSR so soc cpu dose not  need any connection to read or write for example dma buffer (there can be use case although  ))
in summery for now there is no possible way to have a high BW pcielite with soc enabled 
even increasing data wdith wont solve the issue for higher speed US and higher BW at the end the Clock frequncy limitation of 
CPU and interlock of the structure between CPU and pcielite will result in lower clock secltion to make evry thing work 
please find a solution to that or separate pcielite from soc cpu clock doamin properly 
thanks

enjoy-digital, 2022-09-14:
Hi @1n3o1,

as said in https://github.com/enjoy-digital/litepcie/issues/99, please read https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support#feedback--contribution--support and avoid sentence like: "please find a solution to that or separate pcielite from soc cpu clock doamin properly" -> The solution shared/offered covers our needs, if things needs to be extended, it's possible to contribute or see if we could implement it (eventually through a support contract).

For your SoC, the automatic integration offered in LiteX will indeed not allow it directly. But by using [LitePCIe generator ](https://github.com/enjoy-digital/litepcie/blob/master/litepcie/gen.py)and [LiteX SoC generator](https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_soc_gen.py), you can generate separate LitePCIe core/LiteX SoC and re-integrate them in a custom LiteX SoC that will offer you more flexibility regarding clock domains.

closed 2022-09-14 by enjoy-digital

1n3o1, 2022-09-18:
Hi there
the simple solution would be to separate pcie csr form soc csr .
as I said there is no need to have shared csr with soc cpu why would soc cpu need to have access to dma registers(is there any ?) and pcie.
or using CDC on the shared wishbone (that I dont know how or if this is possible though litex environment ).
thanks.