enjoy-digital/litepcie #99: kernel build result in error

1n3o1, 2022-09-03:
thanks in advance 
due to incomplete description of struct type for liteuart.c build result in error 
I had to modify the make file and remove liteuart to be built
there is a missing struct type called struct xa_limit limit;
and also macro XA_LIMIT

enjoy-digital, 2022-09-14:
Hi @1n3o1,

Can your provide the error message?

Also I would recommend reading https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support#feedback--contribution--support before creating an issue: If you don't share the useful information you have (here error message, specific config, etc...), we'll have to spend time trying to guess what you are doing differently.

1n3o1, 2022-09-18:
>Hi
I actuality said it.
missing declaration but here you are :
{
make -C /lib/modules/5.0.0-23-generic/build ARCH=x86_64 CROSS_COMPILE= M=/dir/kernel modules
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-23-generic'
  CC [M]  /dir/kernel/main.o
/dir/kernel/main.c: In function ‘litepcie_pci_probe’:
/dir/kernel/main.c:978:19: warning: unused variable ‘tty_res’ [-Wunused-variable]
  struct resource *tty_res = NULL;
                   ^~~~~~~
  LD [M]  /dir/kernel/litepcie.o
  CC [M]  /dir/kernel/liteuart.o
/dir/kernel/liteuart.c: In function ‘liteuart_probe’:
/dir/kernel/liteuart.c:269:18: error: storage size of ‘limit’ isn’t known
  struct xa_limit limit;
                  ^~~~~
/dir/kernel/liteuart.c:276:11: error: implicit declaration of function ‘XA_LIMIT’; did you mean ‘INT_LIMIT’? [-Werror=implicit-function-declaration]
   limit = XA_LIMIT(0, CONFIG_SERIAL_LITEUART_MAX_PORTS);
           ^~~~~~~~
           INT_LIMIT
/dir/kernel/liteuart.c:284:43: warning: passing argument 3 of ‘xa_alloc’ makes integer from pointer without a cast [-Wint-conversion]
  ret = xa_alloc(&liteuart_array, &dev_id, uart, limit, GFP_KERNEL);
                                           ^~~~
In file included from ./include/linux/radix-tree.h:31:0,
                 from ./include/linux/idr.h:15,
                 from ./include/linux/kernfs.h:14,
                 from ./include/linux/sysfs.h:16,
                 from ./include/linux/kobject.h:20,
                 from ./include/linux/module.h:17,
                 from /dir/kernel/liteuart.c:10:
./include/linux/xarray.h:757:19: note: expected ‘u32 {aka unsigned int}’ but argument is of type ‘struct liteuart_port *’
 static inline int xa_alloc(struct xarray *xa, u32 *id, u32 max, void *entry,
                   ^~~~~~~~
/dir/kernel/liteuart.c:269:18: warning: unused variable ‘limit’ [-Wunused-variable]
  struct xa_limit limit;
                  ^~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:291: recipe for target '/dir/kernel/liteuart.o' failed
make[2]: *** [/dir/kernel/liteuart.o] Error 1
Makefile:1606: recipe for target '_module_/dir/kernel' failed
make[1]: *** [_module_/dir/kernel] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-23-generic'
Makefile:14: recipe for target 'litepcie.ko' failed

}

in short :
error: storage size of ‘limit’ isn’t known
error: implicit declaration of function ‘XA_LIMIT’;

try build the new kernel you will see it.

enjoy-digital, 2022-09-18:
Sorry, I took time to write https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support, so please read it and think about your tone before submitting another issue.

closed 2022-09-18 by enjoy-digital