enjoy-digital/litepcie #110: Support of Non-Transparent Bridge (NTB) for inter-domain communication through PCIe interfaces.

ohault, 2022-12-08:
Do you plan to add support of PCIe Non-Transparent Bridge (NTB) ?

https://www.missinglinkelectronics.com/devzone/index.php/fpga-debug-ntb
https://github.com/jonmason/ntb/wiki

enjoy-digital, 2023-01-05:
This is an interested topic, but a huge project by itself. Such developments are generally done with the help of partners interested and this hasn't been planned yet but we are of course open to discuss to possibilities to would help this this happen :)

ohault, 2023-08-22:
Once the PCIe swich will be [implemented](https://github.com/enjoy-digital/litepcie/issues/117) (using transparent mode bridge), adding non transparent mode would required adding memory address translation windows and isolation between both sides (doorbell registers, as well as non common features like scratchpad and message registers)

enjoy-digital, 2024-01-12:
Hi @ohault,

sorry, but this does not seems to be a realistic feature. That's probably is a > 6 months/developer effort that would need to be funded/discussed together if that something you would like to see happening. I have a preference to only keep realistic features/issues open. If you are interested by this features, please get in touch directly and we'll discuss what can be done.

Regards,

Florent

closed 2024-01-12 by enjoy-digital