enjoy-digital/litepcie #128: PCIe on UltraScalePlus with 256 bit width is broken

smunaut, 2024-02-06:
Commit `526a3d05aafcf8877b7cb9f893de82f3bd85a6e4` breaks it.

Not only does this not account if you have `address_width=32` (see associated PR to fix that issue), but even after that or if you use `address_width=64`, it doesn't work.

The documentation explicitely states  "The user application must keep the s_axis_rq_tvalid signal asserted over the duration of the packet. " but the new code removed the FIFO that was ensuring this.

smunaut, 2024-02-06:
Another issue is with `tkeep`.

It's "per-byte" at the input but at the output it's expected to be "per-double-word". So for one it's too wide ( 32b instead of 8b ) but it's also just plain wrong value ...

This was introduced in the "cleanup" AFAICT.

enjoy-digital, 2024-02-09:
Thanks, I'll spent time next week looking at this and try to get it working with your config. Strangely, I've been using this code in the last months on at least two different systems also on Ultrascale+ and Gen3 X8, so in addition to your finding, there is probably a subtle difference.

smunaut, 2024-02-09:
Factors that help not see the issues :
* `address_width=64`,
* Use only DMAReader (and not DMAWriter),
* sys_clk is fast enough ( >= 250M ) to not cause discontinuity
* Transfers of the exact good length to have tkeep match by "chance"

enjoy-digital, 2024-02-10:
The designs are using DMAReader and DMAWriter but indeed in the "right" combination for the other parameters. I'll merge your PR next week and will do more tests to help/fix things. I'm also setting up a HW CI setup for LiteX to help catching regressions on these wrappers.

commit ab44e172 (2024-02-12): phy/xilinx_{us,usp}: Fix the x8 RQ adaptation layer

When switching over to 4DW format in commit 526a3d05, several things
went broken :

 - The spec explicitely requires that each request is contiguous
   (i.e. tvalid doesn't go down) and that commit removed the FIFO
   that was ensuring that.

 - The tkeep provided at the input of the conversion layer is "per-byte"
   but xilinx wants "per-dword" one.

This commit re-introduces the FIFO and fixes the tkeep generation.

Fixes #128
closed 2024-02-12 by enjoy-digital