enjoy-digital/litepcie #150: Zero Copy mode fails if IOMMU is present

smunaut, 2025-08-09:
I had a machine running 24.04.03 that was failing with some weird message in `dmesg` when attempting to use zero-copy mode.

"[2150502.965087] DMAR: [DMA Read NO_PASID] Request device [02:00.0] fault addr 0xff835000 [fault reason 0x06] PTE Read access is not set"

And this was happening only when Vt-d was enabled in the BIOS ( which in turn, enabled the IOMMU ).

In the end the only way I found to solve it was to use `dma_mmap_coherent` in the `litepcie_mmap` implementation.
However there are a couple of issues with that :
* That call will want to map the whole zone at once, and not small individual buffers ... so I had to change the allocation to use a single dma_alloc_coherent for the whole DMA_BUFFER_TOTAL_SIZE.
* `vma->vm_pgoff` must be set to 0 before calling it since it will use that param to know what offset in the buffer to map ...

In the end I managed to make it work, but not sure if that solution is adequate for upstream or if there is a better solution ?

trabucayre, 2025-08-09:
Could you try: 
```
# Add to GRUB config (/etc/default/grub):
GRUB_CMDLINE_LINUX="iommu=pt"
# or
GRUB_CMDLINE_LINUX="iommu=off"

sudo update-grub && sudo reboot
```

as mentionned [here](https://github.com/enjoy-digital/litex_m2sdr?tab=readme-ov-file#-getting-started) tips section.

trabucayre, 2025-08-12:
Indeed. Driver may be a cause of this issue.
A quick search doesn't provides relevant solutions but it's maybe because pages found are related to ethernet cards or graphics cards without others solutions than changing boot config.

Problem here is to be able to locally reproduces the error.